jgriss/FusionSolarPy #2: Battery Data

rafaeluriarte, 2022-04-24:
I was trying to retrieve the battery percentage but I could not find the data in the "plant_data". It shows only if it has the battery ("existEnergyStore") but not its charge. Do you know if this info is available? In the web interface it is...

jgriss, 2022-04-29:
Hi @rafaeluriarte,

I believe that the charge (for the complete time frame) is stored in `chargePower`. That's where the web interface's graph seems to be getting the data from.

I've updated the `get_last_plant_data` function to also return this value.

Please let me know if it works as expected!

rafaeluriarte, 2022-04-29:
Thank you very much for the effort. Unfortunately, I don't think it is the right variable:

It says: 

Last battery charge at 2022-04-29 21:40: 0.0

But I am at 96%

[image]

Any idea?

Thanks!

jgriss, 2022-05-13:
Hi @rafaeluriarte 

Since I don't have a battery myself, it took a while to figure out. The data for this diagram is part of the energy flow.

I've added a function to get the complete data of this diagram:

```python
energy_flow = client.get_plan_flow(plant_id)
```

It's currently only on GitHub since I still want to add some nicer functions to process this data structure.

rafaeluriarte, 2022-05-14:
Thanks a million, it works like a charm!

The data structure is a bit strange but we can get the battery percentage with
battery = energy_flow['data']['flow']['nodes'][4]['deviceTips']['SOC']

If you need some help with the dev, just let me know

closed 2022-05-14 by rafaeluriarte