jgriss/FusionSolarPy #4: requests.exceptions.HTTPError: 406 Client Error: Not Acceptable for client.get_plant_stats

MichelRabozee, 2022-09-09:
Hi,

I just had installed a photovoltaic plant and I found your project in order to get the data for integrating into my own page.

The "stats = client.get_power_status()" example works fine to retrieve current data, however, when, according to the second example I try to retrieve the "daily" stats data, I get:

```
requests.exceptions.HTTPError: 406 Client Error: Not Acceptable for url: https://region01eu5.fusionsolar.huawei.com/rest/pvms/web/station/v1/overview/energy-balance?stationDn=NE%3D33586185&timeDim=2&queryTime=1662720704354&timeZone=2&timeZoneStr=Europe%2FVienna&_=1662720707250
```

The code correctly gets the one plant I have (through "plant_ids = client.get_plant_ids()", plant_ids[0] gets NE=33586185 as id). The call which fails is "plant_data = client.get_plant_stats(plant_ids[0])"

The FusionSolar app (on iOS ou in a browser) works fine and give me the day stats charts.

What can I do ?

jgriss, 2022-09-17:
Hi @MichelRabozee,

Thanks a lot for reporting this bug! Sorry about the delay! I'll try to fix it ASAP

jgriss, 2022-09-17:
Hi @MichelRabozee,

I did a quick check and the code still works for me.

As a first test: Can you check when you open the FusionSolar web app through a browser, that the URL starts with `region01eu5.fusionsolar.huawei.com`?

Depending on where you are located, the subdomain may be different.

Kind regards,
Johannes

jgriss, 2022-09-17:
Hi,

I just double checked my HomeAssistant log files: The same error occurred once 2 days ago but then never again. Maybe they had some kind of update running on their servers. Right now, I can't reproduce the error and for the last 2 days no additional errors were logged on HomeAssistant as well.

MichelRabozee, 2022-09-17:
Hi,

my kiosk is at https://region01eu5.fusionsolar.huawei.com/pvmswebsite/nologin/assets/build/index.html#/kiosk?kk=xxxxx and my full web app is at https://region01eu5.fusionsolar.huawei.com which goes to https://eu5.fusionsolar.huawei.com/unisso/login.action?service=https%3A%2F%2Fregion01eu5.fusionsolar.huawei.com%2Funisess%2Fv1%2Fauth%3Fservice%3D%252Fnetecowebext%252Fhome%252Findex.html&decision=1#/LOGIN

HTH, and do not hesitate to ask more info (for instance, I can give you my kiosk code in PM so you may check),

kind regards,

Michel

jgriss, 2022-09-17:
Hi,

As a first step, can you maybe simply try whether the code works right now? I just ran all unit tests (which include the code shown in the README) and they worked fine.

MichelRabozee, 2022-09-17:
Hi !
Sorry, still no go :-(
```
MacMini2:fusionsolar mrabozee$ ./get_today_fusionsolar_data.py
Found 1 plants
Found NE=33586185 plant
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 46, in wrapper
    result = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 298, in get_plant_stats
    r.raise_for_status()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 406 Client Error: Not Acceptable for url: https://region01eu5.fusionsolar.huawei.com/rest/pvms/web/station/v1/overview/energy-balance?stationDn=NE%3D33586185&timeDim=2&queryTime=1663450290719&timeZone=2&timeZoneStr=Europe%2FVienna&_=1663450293282

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/mrabozee/sources/fusionsolar/./get_today_fusionsolar_data.py", line 16, in <module>
    plant_data = client.get_plant_stats(plant_ids[0])
  File "/usr/local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 50, in wrapper
    result = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 298, in get_plant_stats
    r.raise_for_status()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 406 Client Error: Not Acceptable for url: https://region01eu5.fusionsolar.huawei.com/rest/pvms/web/station/v1/overview/energy-balance?stationDn=NE%3D33586185&timeDim=2&queryTime=1663450290719&timeZone=2&timeZoneStr=Europe%2FVienna&_=1663450295503
```

The code I am using is:
```
#!/usr/local/opt/python@3.9/bin/python3.9
# -*- coding: UTF-8 -*-

from fusion_solar_py.client import FusionSolarClient

# log into the API - with proper credentials...
client = FusionSolarClient("Michel_Rabozée", "(Matignon14)", huawei_subdomain="region01eu5")

# get the plant ids
plant_ids = client.get_plant_ids()

print(f"Found {len(plant_ids)} plants")
print(f"Found {plant_ids[0]} plant")

# get the data for the first plant
plant_data = client.get_plant_stats(plant_ids[0])

# plant_data is a dict that contains the complete
# usage statistics of the current day. There is
# a helper function available to extract some
# most recent measurements
last_values = client.get_last_plant_data(plant_data)

print(f"Last production at {last_values['productPower']['time']}: {last_values['productPower']['value']}")

# In case you have a battery installed
print(f"Last battery charge at {last_values['chargePower']['time']}: {last_values['chargePower']['value']}")

# log out - just in case
client.log_out()
```

jgriss, 2022-09-18:
Hi @MichelRabozee ,

Just as a warning: You added your username and password in the code example.

I must also apologize: I used them to fix the bug. Your setup is a bit different as your dongle is registered as a separate device. That's why the ID fetching function returned the wrong id.

I've update the package and changed the way the station ids are found (dc8a4ea434d739681e3392e92ea2e579ac00d232). If you install the new version (0.0.7) it should work. Please let me know if it doesn't!

closed 2022-09-18 by jgriss

MichelRabozee, 2022-09-18:
Thank you for having removed the insecure posts. I was in a hurry and completely forgot to obfuscate those :-( I will check the new version a bit later today, I have some issues to attend to.

MichelRabozee, 2022-09-18:
Hello,

seems now to work perfectly ! Thank you !

Have a nice day.