jgriss/FusionSolarPy #5: Failed to login into FusionSolarAPI: Incorrect verification code

MichelRabozee, 2022-09-24:
Hello !

It seems Huawei changed something to their system. Everything was working till 22:00 (European Time) yesterday 23-Sep-2022. From then, when calling client = FusionSolarClient(<user name>, <user pw>, huawei_subdomain="region01eu5"), I get:
```
  File "/usr/local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 82, in __init__
    self._login()
  File "/usr/local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 118, in _login
    raise AuthenticationException(
fusion_solar_py.exceptions.AuthenticationException: Failed to login into FusionSolarAPI: Incorrect verification code. 
```

They seem to have added (for a better client service !!) a Captcha to the authentication scheme :-(

Is it game over ?

Have a nice day,

Michel

jgriss, 2022-09-24:
Hi @MichelRabozee,

Thanks a lot for your feedback!

No, this feature has always been there. Luckily, the validation code is only required if they (seem to) assume some misuse. In my setup, the system has been running for nearly 1 year without an issue. But I keep the session open for many days.

In my experience, the authentication captcha was only requested when I used the wrong password too often (during testing).

Kind regards,
Johannes

MichelRabozee, 2022-09-24:
OK, thank you for the explanations.

So I guess, the Captcha will disappear after a while ? Or do I have to do something ?

Best regards,
Michel

jgriss, 2022-10-02:
Hi @MichelRabozee,

Yes exactly! In my case, the captcha always disappeared again on its own.

Kind regards,
Johannes

closed 2023-06-01 by jgriss

Spiritreader, 2023-08-13:
@FunnyPocketBook and I have made modifications to the log in flow of the library.

It now supports re-using sessions, which reduces the number of login calls that are made if you use the script without a loop (for example as a telegraf plugin.

In addition, we've added a neural network based captcha solver which is automatically used as soon as a captcha is required.

This is available on our fork
https://github.com/Walzen-Group/FusionSolarPy

If desired, we can open a PR.

jgriss, 2023-08-16:
Hi @Spiritreader 

A PR would be great! Thanks a lot!

FunnyPocketBook, 2023-08-16:
Great, thanks for the positive response! We'll do a clean up of the code and write some documentation for the new parameters before submitting the PR.