jgriss/FusionSolarPy #6: Source for calls / API documentation?

rgiessmann, 2022-09-26:
Dear Johannes,

Thanks for the great package! We are having a SUN2000 inverter here, too, and your package seems to allow access to the data without applying for an openAPI access at Huawei first. Great!

I was wondering how you derived the API call patterns; I couldn't find any documentation in the internet. Are there maybe more calls available, which I could help to implement? 

Thanks and best,
   Robert

jgriss, 2022-10-02:
Hi @rgiessmann,

Thanks  a lot for the very nice feedback!

Essentially, I am using Chrome's developer tools to debug the API and rebuild it in the package. So in other words the package tries to immitate the web application - which is why one doesn't need the openAPI access. 

As you probably saw, I focused on the calls that are most relevant to my own setup.

It would be great if you could contribute additional functions / features to the code.

Kind regards,
Johannes

rgiessmann, 2022-10-03:
Hi @jgriss ,

Thanks for letting me know. I asked the Huawei support for the API documentation, but only got the "openAPI" one. 

I hope to be able to contribute soon. 

Indeed, the "doesn't need openAPI access" is the big advantage of this package! 

Happy to speak to you!

Best,
   Robert

closed 2022-10-03 by rgiessmann