jgriss/FusionSolarPy #10: Question regarding login procedure

reyntjensw, 2023-05-15:
Hi

I'm writing a FusionSolar plugin for the Homey platform but I'm running into issue regarding the API (which sometimes work and sometimes is not). That is why I'd like to use the way you authenticate and get your data. But for this to work I need to convert your Python code to Node code but there I have an issue and maybe you encountered that in the past already. Any thoughts would be helpful.

So the first and the most easy call works fine, I get a result https://github.com/jgriss/FusionSolarPy/blob/main/src/fusion_solar_py/client.py#L100.

But on the second call I do not succeed in retrieving any data https://github.com/jgriss/FusionSolarPy/blob/main/src/fusion_solar_py/client.py#L123, where your script does return data.
The only return for the second call is an empty fetch object with some headers.

If you could give me any insights that would be wonderful.

Thanks

` const login_url = "https://" + url.slice(-3) + ".fusionsolar.huawei.com/unisso/v2/validateUser.action?"

        const params = new URLSearchParams({})
        params.append("decision", 1)
        params.append("service", 'https://region01eu5.fusionsolar.huawei.com/unisess/v1/auth?service=/netecowebext/home/index.html#/LOGIN')

        const apiResponse = await fetch(login_url + params.toString(), {
            method: 'POST',
            headers: {
                'Content-Type': 'application/json'
            },
            body: JSON.stringify({
                "organizationName": "",
                "username": this.username,
                "password": this.password
            })

        });

        const data = await apiResponse.json();
        console.log(data)

        if (data.status !== 200) {
            // throw new console.error("Login failed");
        }

        const params2 = new URLSearchParams({})
        params2.append("_", Math.round(Date.now()))
        console.log(params2);
        console.log(this.server)
        const session = await fetch("https://" + this.server + ".fusionsolar.huawei.com/rest/neteco/web/organization/v2/company/current?" + params2.toString(), {
            method: "GET"
        })

        const sessionOutput = await session;
        console.log(sessionOutput);`

jgriss, 2023-05-16:
Hi @reyntjensw 

I'm not 100% I understand your code correctly. In my tool, all requests are send through a session - which takes care of handling all cookies etc. as well as redirects. Is this taken care of in your code?

reyntjensw, 2023-05-17:
hi @jgriss thank for your response, I think I do but I'll take a look

reyntjensw, 2023-05-19:
@jgriss I managed to make it kind of work, there is still some work to be done.
One last question do you have any idea how I can access these values? PV power, I already have, but the current values, I was not able to find. (general values like yield today, total yield, ... ) were easy to find.
[image]

Thanks

jgriss, 2023-06-01:
Hi @reyntjensw ,

Yes - they are found in the "plant flow".

The end-point for this is: [...] fusionsolar.huawei.com/rest/pvms/web/station/v1/overview/energy-flow

The returned data structure is a bit strange but the data is there.

reyntjensw, 2023-06-01:
Great thanks!

closed 2023-06-01 by reyntjensw