jgriss/FusionSolarPy #18: CAPTCHA libraries not available after installing it

ATNALDC, 2023-12-01:
I've updated to version 0.0.16 and using pip install fusion-solar-py[captcha] but when I launch it I get the following error:

`Required libraries for CAPTCHA solving are not available. Please install the package using pip install funsion_solar_py[captcha].
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 124, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 287, in _login
    raise CaptchaRequiredException("Login failed: Incorrect verification code.")
fusion_solar_py.exceptions.CaptchaRequiredException: Login failed: Incorrect verification code.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/captcha_solver_onnx.py", line 4, in <module>
    import cv2
  File "/home/pi/.local/lib/python3.9/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/home/pi/.local/lib/python3.9/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/pi/.local/lib/python3.9/site-packages/cv2/cv2.abi3.so: undefined symbol: __atomic_store_8

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/etc/openhab/scripts/huawei_web.py", line 13, in <module>
    client = FusionSolarClient("ATNALDC", "Aldeanuevasanz30", huawei_subdomain="region03eu5", captcha_model_path="/etc/openhab/scripts/captcha_huawei.onnx")
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 191, in __init__
    self._configure_session()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 298, in _configure_session
    self._login()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 132, in wrapper
    captcha_present = self._check_captcha()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 229, in _check_captcha
    self._init_solver()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 254, in _init_solver
    from .captcha_solver_onnx import Solver
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/captcha_solver_onnx.py", line 9, in <module>
    raise FusionSolarException("Required libraries for CAPTCHA solving are not available. Please install the package using pip install funsion_solar_py[captcha].")
`

It keeps saying that libraries are not  available. 

Could someone tell me what I could be doing wrong?

Thank you very much,
Kind regards,

jgriss, 2023-12-05:
Hi @ATNALDC 

Thanks a lot for the feedback!

It seems that you posted your username and password above. You may want to change that (I edited your comment to remove them).

Quickly double checking your code on a fresh install everything worked fine.

The issue seems to be here:

```
ImportError: /home/pi/.local/lib/python3.9/site-packages/cv2/cv2.abi3.so: undefined symbol: __atomic_store_8
```

For some reason, your open cv2 installation does not seem to work.

To quickly test this, I suggest to create a `venv`, re-install the fusion_solar_py there and test the code.

ATNALDC, 2023-12-14:
Thank you very much @jgriss 

Finally I solved it reinstalling all the libraries.

closed 2023-12-14 by ATNALDC