jgriss/FusionSolarPy #23: Issue after Huawei's server update

ATNALDC, 2024-01-21:
Hi,

The server of huaweis was updated yerterday and move to a uni...eu5 one. I have also the 0.0.20 version but i'm getting this error:

`Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 6 column 1 (char 5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/etc/openhab/scripts/huawei_web.py", line 13, in <module>
    client = FusionSolarClient("XXXXXX", "XXXXXX", huawei_subdomain="uni003eu5", captcha_model_path="/etc/openhab/scripts/captcha_huawei.onnx")
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 232, in __init__
    self._configure_session()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 385, in _configure_session
    self._session.headers["roarand"] = r.json()[
  File "/home/pi/.local/lib/python3.9/site-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 6 column 1 (char 5)`

Can some one help me? 

Thank you very much,
Kind regards,

jgriss, 2024-01-21:
Hi @ATNALDC 

Thanks a lot for reporting this. Apparently this (known) issue triggers other exceptions depending on how you use the package.

Should be fixed now in version v0.0.21 (ad54570813c4663dccc27c29bee9768506a6f201). Please let me know it if isn't!

closed 2024-01-21 by jgriss

ATNALDC, 2024-01-21:
Hi @jgriss 

Now it seems to log in but i get the following error related to get_power_status

`Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 6 column 1 (char 5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/etc/openhab/scripts/huawei_web.py", line 22, in <module>
    stats = client.get_power_status()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 144, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 408, in get_power_status
    power_obj = r.json()
  File "/home/pi/.local/lib/python3.9/site-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 6 column 1 (char 5)
`

Thank you very much,
Kind regards

ATNALDC, 2024-01-21:
Hi again,

After some checks I have seen that this issue only occurs when ussing the session in FusionSolarClient call:

`session2 = requests.Session()

with open('/etc/openhab/scripts/session_huawei.plk', 'rb') as f:
	session2 = pickle.load(f)

client = FusionSolarClient("XXXX", "XXXXX", huawei_subdomain="uni003eu5", captcha_model_path="/etc/openhab/scripts/captcha_huawei.onnx", session = session2)

# get the stats
stats = client.get_power_status()

with open('/etc/openhab/scripts/session_huawei.plk', 'wb') as f:
	pickle.dump(session2, f)

# log out - just in case
client.log_out()`

If I not use the session option it works.

Other thing I have seen is that "coal", "co2", "coalyear" and "co2year" has been deprecated?

Thank you very much,
Kind regards,

jgriss, 2024-01-21:
Hi @ATNALDC 

Unfortunately, my account was not yet moved to the new unixxx API. Therefore, I cannot debug the behavior.

In a quick test with my credentials, everything still worked.

ATNALDC, 2024-01-23:
Hi @jgriss 

After some research I found, that ussing the code to reuse session produce an error on the login if it's the first time you use it and, the file with session information is empty. In this case the file to store session can't be set because _configure_session() it's only call if session doesn't exists. and during the whole process you use a "fake" session which doesn't let you to log in. 

Maybe, as suggestion could be interesting to have a method to be able to read the session information, so you can write it into the session file the first time you use the code, and after that you can then set the code as specified on the readme.

Thank you very much!
kind regards