jgriss/FusionSolarPy #24: Weird data Received from the new subdomains

mtziortzis, 2024-02-11:
First off, thanks for this great library. I was using this just fine until i got migrated to the new subdomain. The data i receive now are a bit off.
I verified:
- i receive the right plant id (double checked with the url while i am logged in)
- i have the right subdomain (new one)

Here is the response i am getting from the

remedix, 2024-02-17:
any ideas @jgriss  ? :)

mtziortzis, 2024-03-11:
Hi @jgriss 
Thank you for taking some time to look into this.

I think it comes down to state results. Let me explain.

In the current dashboard of fusionsolar there are two sets of data. The Plant KPIs (which seems to be off (delayed?) as shown here:
[image]

And the almost realtime data shown here:
[image]

The method `fsclient.get_power_status()` gets the first data (Plant KPIs) - the 'stale' ones

However the API endpoint `uni003eu5.fusionsolar.huawei.com/rest/pvms/web/station/v1/overview/station-real-kpi` seems to have the latest (real time) ones.

I tried the following from the browser
- clear cache
- clear cookies
- clear local storage

Nothing seems to be changing the stale data so they must be coming from the server. 

I think `fsclient.get_power_status()` should get the data from the endpoint as explained above - this will ensure the latest data. I hope this steers you to the right direction.

jgriss, 2024-06-22:
Hi @mtziortzis ,

Sorry, time available for the project is a bit irregular...

Your observation is correct, the numbers shown in the overview seem to be stale.

Yet, you can already get the up-to-date data shown in the table using the `get_plant_stats` function combined with `get_last_plant_data`.

There, the `productPower` slot corresponds to the current production value.

jgriss, 2024-06-22:
Sorry, just read the first part of your message. In my tests, the values shown in the table and the ones in get_last_plant data match. I'll check how I can support the new endpoint though

jgriss, 2024-06-22:
Hi,

A second update: The more up-to-date data is also available in the station list which you can retrieve using `get_station_list`

jgriss, 2024-06-22:
@mtziortzis I've added a new function `get_current_plant_data` in version 0.0.24 that uses the new endpoint.

commit 4bca92b5 (2024-06-22): Fixes #24: Added new function get_current_plant_data
closed 2024-06-22 by jgriss