jgriss/FusionSolarPy #25: Error trying to reach the data from another days

Jongonzale, 2024-02-26:
Hi,

Until now I was reaching the info of the generations of previous days without any problem. But after 2 weeks without using it, now when I try to reach that information (the one that gives the information every 5 minutes, the productPower) if I try to reach todays information it works without problems, but when I try with a previous day, I reach only "--" in every one. I debug it and I can also see that the totalProductPower is good, it's only in that one, the productPower

jgriss, 2024-03-08:
Hi @Jongonzale ,

Thanks a lot for reaching out!

How did you set the `query_time` parameter in your call?

Jongonzale, 2024-03-12:
Hi @jgriss 

This is what I do to call the client.get_plants_stats:
`try:
            plant_ids = client.get_plant_ids()
            query_time = client._get_day_start_sec()
            query_time = query_time - days * 86400
            plant_data = client.get_plant_stats(plant_ids[0],query_time)
            # last_values = client.get_last_plant_data(plant_data)
            # stats = client.get_power_status()
            break
        except Exception as e:
            # print("Error al conseguir stats: ")
            intentosC = intentosC + 1
            time.sleep(5)
   return plant_data['productPower']
`

And I call the method with the days variable. If I set that variable as 0, the list I obtain is a good one (an example: ', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '0.000', '0.000', '0.000', '0.000', '0.000', '0.000', '0.000', '0.000', '0.000', '0.307', '0.393', '0.352', '0.364', '0.382', '0.459', '0.496', '0.667', '0.516', '0.792', '0.903', '1.021', '1.043', '1.656', '2.401', '1.876', '2.345', '1.616', '3.123', '2.720', '2.910', '2.432', '2.297', '2.777', '2.818', '4.427', '4.562', '3.260', '3.300', '3.355', '4.368', '5.455', '4.108', '5.655', '7.131', '8.778', '6.463', '4.953', '3.954', '7.216', '8.981', '8.062', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', 
'--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--']) . But If I set the days variable as >0 then I obtain this: ['--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--', '--']

Jongonzale, 2024-03-19:
Hi @jgriss ,

I'm still trying with different options and I'm not able to get the day resume information. The totalProductPower of the previous days is good, but the problems comes with the productPower resume. I need that information for getting the info of some days ago. Do you know why it can be?

Thanks

jgriss, 2024-06-22:
Hi,

I use a different code to set the query time:

```
start_today = time.strftime("%Y-%m-%d 00:00:00", time.gmtime())
struct_time = time.strptime(start_today, "%Y-%m-%d %H:%M:%S")
seconds = round(time.mktime(struct_time) * 1000)
```

I'd suggest you try this approach and simply adapt the date. This seems to work in my tests.

Jongonzale, 2024-07-24:
> Hi,
>
> I use a different code to set the query time:
>
> ```
> start_today = time.strftime("%Y-%m-%d 00:00:00", time.gmtime())
> struct_time = time.strptime(start_today, "%Y-%m-%d %H:%M:%S")
> seconds = [...]

Hi,

I used this way for eaching information from other days and like this it works for me. I don't know why the previous way it doesn't work, but with this is it okey so perfect.

Thanks

Jongonzale, 2024-07-24:
> Hi,
>
> I use a different code to set the query time:
>
> ```
> start_today = time.strftime("%Y-%m-%d 00:00:00", time.gmtime())
> struct_time = time.strptime(start_today, "%Y-%m-%d %H:%M:%S")
> seconds = [...]

Hi,

I tried to reach the information from other days using your way and it works for me. I don't know why the previous way fails, but with yours it works perfectly.

Thanks

closed 2024-07-24 by Jongonzale