jgriss/FusionSolarPy #26: Error when running a script continously

Hazkarax, 2024-06-07:
I am trying to continuously run a script to provide live production data of a solar plant, which is to be displayed on a raspberry which will run continuously. After approximately 20-30 minutes, its stops updating, which I assume is due to the API returning a captcha. 

I have tried to resolve this error both by re-using the session and by using captcha. However, the captcha ad-on cannot be installed as pip can not find any matching packages. Running the code with the captcha config therefore returns an error after 20 minutes when the captcha is needed.

I do not quite understand how to re-use the session either, as it gives an error when it tries to read a json. How does a full implementation of a session re-use look like?

The code in which I am trying to implement the following looks like this. Adding the recaptcha to the client gives an error after 20 minutes when a verification is requested, and the implementations I have tried of a session re-use have provided an error when reading the json.

```python
from fusion_solar_py.client import FusionSolarClient
import pickle
import requests

session = requests.Session()
client = FusionSolarClient(
    'username',
    'password',
    huawei_subdomain="uni004eu5"
)

def perform_login():
    plant_ids = client.get_plant_ids()
    return plant_ids

def get_solar_power(plant_ids):
    

    for plant in range(len(plant_ids)):
        if plant_ids[plant] == "specific_plant_id":
            plant_data = client.get_plant_stats(plant_ids[plant])
            last_values = client.get_last_plant_data(plant_data)
            return str(last_values['productPower']['value'])

def get_solar_production(plant_ids):

    for plant in range(len(plant_ids)):
        if plant_ids[plant] == "specific_plant_id":
            plant_data = client.get_plant_stats(plant_ids[plant])
            last_values = client.get_last_plant_data(plant_data)
            return str(last_values['totalUsePower'])
        

```

jgriss, 2024-06-22:
Hi @Hazkarax,

In your code, you are currently not really using the session object you create. This would have to be passed through the `session` parameter in the constructor.

In my experience, the easiest way to keep an application running for a long time (in my case nearly 2 years without a CAPTCHA test) is by limiting the number of logins to a minimum.

I've updated the README.md and added instructions how to use two new calls to ensure that a session is kept alive. Essentially, you need to call `is_session_active` around every 10 seconds, and `keep_alive` every 30 seconds.

In my case, this allowed me to use the same session over many months.

Hazkarax, 2024-06-24:
Thanks for your reply. I have tried to update the code, and this is what happens.

When I run this version of the code:

```python
from fusion_solar_py.client import FusionSolarClient
import requests

# log into the API - with proper credentials...
session = requests.Session()
client = FusionSolarClient(
    ‘Username’,
    ‘Password’,
    huawei_subdomain="uni004eu5",
    session=session
)

plant_ids = client.get_plant_ids()

print(f"Found {len(plant_ids)} plants")

```

I get this error message.

```
File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/fusion_solar_py/client.py", line 456, in get_station_list
    obj_tree = r.json()

requests.exceptions.JSONDecodeError: Expecting value: line 6 column 1 (char 5)
```

At r.json, the API  doesn’t return a json but a HTML-page which appears to be a login page, and looks like this:
[image]

Hazkarax, 2024-09-17:
Hi @jgriss!

Have you had any chance to look at this? Is there something wrong from my side based on my previous reply?

jgriss, 2024-09-17:
Hi @Hazkarax ,

This seems to be related to the new encrypted login procedure required by the API (see #30)

This was fixed in v0.0.26

Could you check whether you updated to the latest version?

closed 2025-03-15 by jgriss