jgriss/FusionSolarPy #27: It seems that some URLs have change

ATNALDC, 2024-06-30:
Hi,

Since yesterday the I get some errors. After investigatin the origin I have seen that the web has been changed.

For example if I go to the url used to get plant list (https://uni003eu5.fusionsolar.huawei.com/rest/pvms/web/station/v1/station/station-list) I get 404 error. It also happens with device list.

Could you please check it?

Thank you very much;
Best regards,

karstenBriksoft, 2024-07-20:
I’m not sure if this is a problem with logging in or with the subdomain. But on the webpage the urls seem to work, but they use different subdomains than the python code.

jgriss, 2024-09-17:
Hi,

Some issues were related to the new encrypted login procedure (see #30)

This was solved in the most recent version (v0.0.26)

Getting the station list works in my unit tests. Could you check whether the error persists?

ATNALDC, 2024-11-08:
It's solved with latest version.

closed 2024-11-08 by ATNALDC