jgriss/FusionSolarPy #28: Authentication Failure

IvRogoz, 2024-07-01:
Since middle of last week Im unable to authenticate on Fusion.

AuthenticationException: Failed to login into FusionSolarAPI: None

Any idea why?

denimmonkey, 2024-07-01:
I am having the same issue since the last week 

My domain is also 'uni001eu5'

denimmonkey, 2024-07-01:
The error comes  for the function get_station_list, get_plant_stats still works 
```
ERROR: Failed to reset session and login again.
Login apparently failed. Received invalid response.

```

Jongonzale, 2024-07-01:
In my case I'm getting the error also with the get_plant_ids method

fvillemin, 2024-07-03:
Same error here, plant stats are available but other methods give
Login apparently failed. Received invalid response.

thematt87, 2024-07-05:
I get also 470 error code on login on `uni002eu5`

Jongonzale, 2024-07-09:
How you can know the plant id without the get_plant_ids method? Some said that it's that method the one that fails but that you can reach the info with other methods. I need the id

jgriss, 2024-08-05:
Hi! This is the same issue as #29 

I'll update #29 with more information

commit 689969f0 (2024-08-11): Fixes #28 and #29 Added support for login function v3

The login procedure now supports encrypted passwords which are required by the login function of API version 3
closed 2024-08-11 by jgriss