jgriss/FusionSolarPy #29: Failed to reset session and login again.

JesusRosaB, 2024-07-02:
I am obtaining this error after create the client, some idea?

JesusRosaB, 2024-07-03:
> i get it when i call **get_plant_ids** or **get_station_list** , but not for **get_station_list** . There is a problem in: `fusion_solar_py/client.py", line 551, in get_station_list obj_tree = r.json()`
> 
> the response is an html not json.

But you obtain same error? i thought was related with login process not with that methods

thematt87, 2024-07-05:
I'm having the same issue with the following code from the example:
`from fusion_solar_py.client import FusionSolarClient

client = FusionSolarClient("xxx", "xxx", huawei_subdomain="uni002eu5")

plant_overview = client.get_station_list()
print(f"Current power production: { plant_overview[0]['currentPowwer'] }")`

I think Huawei changed somehting in the authentication, because if I capture the REST call from the Fusion Solar Web Portal and execute it with the fake-cookie from Chrome, the API works.

Anyone an idea how to fix this?

JesusRosaB, 2024-07-05:
> I'm having the same issue with the following code from the example:
> 
> ```
> from fusion_solar_py.client import FusionSolarClient
> 
> client = FusionSolarClient("xxx", "xxx", huawei_subdomain="uni002eu5")
> 
> plant_overview = client.get_station_list()
> print(f"Current power production: { plant_overview[0]['currentPower'] }")
> ```
> 
> I think Huawei changed somehting in the authentication, because if I capture the REST call from the Fusion Solar Web Portal and execute it with the fake-cookie from Chrome, the API works.
> 
> Anyone an idea how to fix this?

As said @animatrons the problem is "get_station_list()" method, if you try to avoid its use should work your code

Jongonzale, 2024-07-09:
> As said @animatrons the problem is "get_station_list()" method, if you try to avoid its use should work your code

And how do you know the plant_id without that method or the get_plant_ids?

JesusRosaB, 2024-07-09:
> And how do you know the plant_id without that method or the get_plant_ids?

i had the IDs in my own DataBase, then i hardcoded the ids in the code until method its not working

thematt87, 2024-07-09:
> i had the IDs in my own DataBase, then i hardcoded the ids in the code until method its not working

That is not a solution for me. I don't know my IDs and I also don't find them in the Fusionsolar web portal :(

Jongonzale, 2024-07-09:
> That is not a solution for me. I don't know my IDs and I also don't find them in the Fusionsolar web portal :(

I see what's my ID. WHta I've done is this:
-Go to the Fusion Solar portal to the 'Control' part.
-There see with the DevTools  (push the F12) the info of the web
-Go to the Network part and refresh the web
-You'll see in the left part one with the name 'energy-balance?stationDn...'
-Click that and copy the Request URL in another internet page
-You'll see the JSON it returns, and the ID is the 'stationDn' one.

I think that there would be an easier way, but that's the first way I find it and works :)

RRonGit, 2024-07-12:
Hardcoding the ID into the code works for me.

thematt87, 2024-07-22:
Thanks @Jongonzale, the workaround does the job for me.

dogganos, 2024-08-04:
Will there be a fix at sometime?

jgriss, 2024-08-05:
Hello everyone!

Sorry about the late response!

The reason for this is that the v2 authentication method used by the library was apparently disabled. The new v3 version uses encrypted passwords. These depend on a public key together with a yet unknown salt.

I still need to identify the code for their new login procedure in order to update the tool.

jgriss, 2024-08-11:
Hi everyone!

This is now fixed in the new version 0.0.25

In case you still experience any issues, please open a new issue.

closed 2024-08-11 by jgriss