jgriss/FusionSolarPy #30: Unable to get plant IDs with message "Failed to reset session and login again"

dogganos, 2024-08-12:
I upgraded to 0.0.25 after https://github.com/jgriss/FusionSolarPy/issues/29 was solved and I still get the same problem. Actually in the beginning I was getting an error about some public key, I then upgraded all required packages and then returned to the "Failed to reset session and login again" message.

However, to make a proper bug report, I fired a clean, live debian 12, and installed fusion_solar_py, but I get the same reset session problem.

Here are the module versions installed:

Requirement already satisfied: fusion_solar_py in /usr/local/lib/python3.11/dist-packages (0.0.25)
Requirement already satisfied: requests>=2.31.0 in /usr/local/lib/python3.11/dist-packages (from fusion_solar_py) (2.32.3)
Requirement already satisfied: cryptography in /usr/local/lib/python3.11/dist-packages (from fusion_solar_py) (43.0.0)
Requirement already satisfied: charset-normalizer<4,>=2 in /usr/lib/python3/dist-packages (from requests>=2.31.0->fusion_solar_py) (3.0.1)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests>=2.31.0->fusion_solar_py) (3.3)
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/lib/python3/dist-packages (from requests>=2.31.0->fusion_solar_py) (1.26.12)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests>=2.31.0->fusion_solar_py) (2022.9.24)
Requirement already satisfied: cffi>=1.12 in /usr/local/lib/python3.11/dist-packages (from cryptography->fusion_solar_py) (1.17.0)
Requirement already satisfied: pycparser in /usr/local/lib/python3.11/dist-packages (from cffi>=1.12->cryptography->fusion_solar_py) (2.22)

And here is the code:
client = FusionSolarClient("XXXX", "XXXX", huawei_subdomain="uni003eu5") # I also tried it with region...
plant_ids = client.get_plant_ids()

This get an exception at the get_plant_ids() line with exception message "Failed to reset session and login again".

jgriss, 2024-08-13:
Hi @dogganos 

Thanks a lot for reporting this!

Could you please post the complete stack trace from the exception? Ideally even the log file from the application.

You can enable complete logging with the following code just before you call the `client = FusionSolarClient...` line.

```python3
import logging

logging.basicConfig(level=logging.DEBUG, filename="fusion_solar_py.log")
```

Then, all log messages are written to the `fusion_solar_py.log` file which you could post here.

dogganos, 2024-08-13:
Hi! 
Here is the complete stack trace of the exception:
```
Login apparently failed. Received invalid response.
Traceback (most recent call last):
File "/home/user/venv/lib/python3.11/site-packages/requests/models.py", line 974, in json
return complexjson.loads(self.text, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
return _default_decoder.decode(s)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
obj, end = self.raw_decode(s, idx=_w(s, 0).end())
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 6 column 1 (char 5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/user/venv/lib/python3.11/site-packages/fusion_solar_py/client.py", line 150, in wrapper
result = func(self, *args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/user/venv/lib/python3.11/site-packages/fusion_solar_py/client.py", line 577, in get_station_list
obj_tree = r.json()
^^^^^^^^
File "/home/user/venv/lib/python3.11/site-packages/requests/models.py", line 978, in json
raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 6 column 1 (char 5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/user/./getDailyProduction.py", line 47, in <module>
plant_ids = client.get_plant_ids()
^^^^^^^^^^^^^^^^^^^^^^
File "/home/user/venv/lib/python3.11/site-packages/fusion_solar_py/client.py", line 150, in wrapper
result = func(self, *args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/user/venv/lib/python3.11/site-packages/fusion_solar_py/client.py", line 547, in get_plant_ids
station_list = self.get_station_list()
^^^^^^^^^^^^^^^^^^^^^^^
File "/home/user/venv/lib/python3.11/site-packages/fusion_solar_py/client.py", line 154, in wrapper
raise FusionSolarException("Failed to reset session and login again.")
fusion_solar_py.exceptions.FusionSolarException: Failed to reset session and login again.
```

And here is the debug log you asked for (I hope it doesnt expose my credentials in some reversible form):

```
DEBUG:fusion_solar_py.client:Logging into Huawei Fusion Solar API
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): eu5.fusionsolar.huawei.com:443
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /unisso/pubkey HTTP/11" 200 None
DEBUG:fusion_solar_py.client:Using V3 loging function with encrypted passwords
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "POST /unisso/v3/validateUser.action?timeStamp=1723538346522&nonce=487b7209f41346f771ae4afc2eb37432 HTTP/11" 200 None
DEBUG:fusion_solar_py.client:New loging procedure successful, sending additional request
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /rest/dp/web/v1/auth/on-sso-credential-ready?ticket=ST-1081465-ifJHQAsyXPRWAP13VNCwbggdDJBSvbl6OG5&regionName=region003 HTTP/11" 302 0
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): uni003eu5.fusionsolar.huawei.com:443
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /uniportal/pvmswebsite/assets/build/cloud.html?app-id=smartpvms&instance-id=smartpvms&zone-id=region-3-55eed080-a3ec-4a37-a366-3049a2747a09 HTTP/11" 200 559
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): region03eu5.fusionsolar.huawei.com:443
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "GET /rest/neteco/web/organization/v2/company/current?_=1723538346683 HTTP/11" 200 153
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "GET /unisess/v1/auth/session HTTP/11" 302 0
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "GET /unisess/v1/auth?service=%2Funisess%2Fv1%2Fauth%2Fsession HTTP/11" 302 0
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "GET /unisso/login.action?service=%2Funisess%2Fv1%2Fauth%3Fservice%3D%252Funisess%252Fv1%252Fauth%252Fsession&decision=1 HTTP/11" 302 22
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /unisso/login.action HTTP/11" 200 None
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "GET /rest/dpcloud/auth/v1/is-session-alive HTTP/11" 200 None
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "GET /rest/dpcloud/auth/v1/is-session-alive HTTP/11" 200 None
DEBUG:urllib3.connectionpool:https://region03eu5.fusionsolar.huawei.com:443 "POST /rest/pvms/web/station/v1/station/station-list HTTP/11" 302 22
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /unisso/login.action HTTP/11" 200 None
ERROR:fusion_solar_py.client:Login apparently failed. Received invalid response.
```

dogganos, 2024-08-13:
And this is if I use huawei_subdomain="uni003eu5" instead of huawei_subdomain="region03eu5":

```
DEBUG:fusion_solar_py.client:Logging into Huawei Fusion Solar API
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): eu5.fusionsolar.huawei.com:443
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /unisso/pubkey HTTP/11" 200 None
DEBUG:fusion_solar_py.client:Using V3 loging function with encrypted passwords
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "POST /unisso/v3/validateUser.action?timeStamp=1723538579947&nonce=8c3361cb57cd7dd159735d207bf22255 HTTP/11" 200 None
DEBUG:fusion_solar_py.client:New loging procedure successful, sending additional request
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /rest/dp/web/v1/auth/on-sso-credential-ready?ticket=ST-1077668-CDXPaWS92Y3j7zlNaxZXlESopwSp4vgnqwa&regionName=region003 HTTP/11" 302 0
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): uni003eu5.fusionsolar.huawei.com:443
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /uniportal/pvmswebsite/assets/build/cloud.html?app-id=smartpvms&instance-id=smartpvms&zone-id=region-3-55eed080-a3ec-4a37-a366-3049a2747a09 HTTP/11" 200 559
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /rest/neteco/web/organization/v2/company/current?_=1723538580131 HTTP/11" 200 153
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /unisess/v1/auth/session HTTP/11" 302 0
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /unisess/v1/auth?service=%2Funisess%2Fv1%2Fauth%2Fsession HTTP/11" 302 0
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /unisso/login.action?service=%2Funisess%2Fv1%2Fauth%3Fservice%3D%252Funisess%252Fv1%252Fauth%252Fsession&decision=1 HTTP/11" 302 22
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /unisso/login.action HTTP/11" 200 None
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /rest/dpcloud/auth/v1/is-session-alive HTTP/11" 200 None
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "GET /rest/dpcloud/auth/v1/is-session-alive HTTP/11" 200 None
DEBUG:urllib3.connectionpool:https://uni003eu5.fusionsolar.huawei.com:443 "POST /rest/pvms/web/station/v1/station/station-list HTTP/11" 302 22
DEBUG:urllib3.connectionpool:https://eu5.fusionsolar.huawei.com:443 "GET /unisso/login.action HTTP/11" 200 None
ERROR:fusion_solar_py.client:Login apparently failed. Received invalid response.
```

jgriss, 2024-08-13:
Hi @dogganos 

Thanks a lot for the detailed information! Apparently, a new header was introduced that's evaluated by some of the function calls.

Should be fixed in v0.0.26 (will be available within the next 30 min.

If the error persists, please open a new issue.

closed 2024-08-13 by jgriss

dogganos, 2024-08-14:
It worked, you are the best, thanks!

commit 929b9231 (2024-08-13): Fixes #30 Added support for roarand header

This header is required for certain calls and contains the payload returned from keep-alive.