jgriss/FusionSolarPy #31: Issue when loggin

ATNALDC, 2024-09-05:
After updating to version 0.0.26 I haven't been able to loggin. I get the following error:

`Failed to load public key.
'function' object has no attribute 'load_pem_public_key'
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/encryption.py", line 53, in encrypt_password
    public_key = serialization.load_pem_public_key(
  File "/usr/lib/python3/dist-packages/cryptography/hazmat/primitives/serialization/base.py", line 23, in load_pem_public_key
    return backend.load_pem_public_key(data)
AttributeError: 'function' object has no attribute 'load_pem_public_key'
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/encryption.py", line 53, in encrypt_password
    public_key = serialization.load_pem_public_key(
  File "/usr/lib/python3/dist-packages/cryptography/hazmat/primitives/serialization/base.py", line 23, in load_pem_public_key
    return backend.load_pem_public_key(data)
AttributeError: 'function' object has no attribute 'load_pem_public_key'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/etc/openhab/scripts/huawei_test_basic.py", line 4, in <module>
    client = FusionSolarClient("XXXXX", "XXXXXX")
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 238, in __init__
    self._configure_session()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 387, in _configure_session
    self._login()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 169, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 327, in _login
    password = encrypt_password(key_data=key_data, password=password)
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/encryption.py", line 60, in encrypt_password
    raise FusionSolarException("Failed to load public key for encryption.")
fusion_solar_py.exceptions.FusionSolarException: Failed to load public key for encryption.
`

Do you know what it could be happening?
Thank you very much,
Kind regards,

jgriss, 2024-09-17:
Hi @ATNALDC 

I tried to replicate your error but failed. In my unit tests and also my own system at home the login is working as expected. Does the error still persist?

Also, could you check that your version of `cryptography` is up to date?

ATNALDC, 2024-09-25:
Hi @jgriss,

I have no yet try to install last version again. I hope to be able to do it this wekeen. I will let you know.

Thank you very much,

ATNALDC, 2024-11-08:
Hi @jgriss 

Sorry for the delay.

I've already install the last version but I get the same error:

`Failed to load public key.
'function' object has no attribute 'load_pem_public_key'
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/encryption.py", line 53, in encrypt_password
    public_key = serialization.load_pem_public_key(
  File "/usr/lib/python3/dist-packages/cryptography/hazmat/primitives/serialization/base.py", line 23, in load_pem_public_key
    return backend.load_pem_public_key(data)
AttributeError: 'function' object has no attribute 'load_pem_public_key'
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/encryption.py", line 53, in encrypt_password
    public_key = serialization.load_pem_public_key(
  File "/usr/lib/python3/dist-packages/cryptography/hazmat/primitives/serialization/base.py", line 23, in load_pem_public_key
    return backend.load_pem_public_key(data)
AttributeError: 'function' object has no attribute 'load_pem_public_key'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/etc/openhab/scripts/huawei_web.py", line 13, in <module>
    client = FusionSolarClient("XXXXXX", "XXXXXX", huawei_subdomain="uni003eu5", captcha_model_path="/etc/openhab/scripts/captcha_huawei.onnx")
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 238, in __init__
    self._configure_session()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 387, in _configure_session
    self._login()
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 169, in wrapper
    result = func(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/client.py", line 327, in _login
    password = encrypt_password(key_data=key_data, password=password)
  File "/home/pi/.local/lib/python3.9/site-packages/fusion_solar_py/encryption.py", line 60, in encrypt_password
    raise FusionSolarException("Failed to load public key for encryption.")
fusion_solar_py.exceptions.FusionSolarException: Failed to load public key for encryption.
`

ATNALDC, 2024-11-08:
Hi again,

I finally found, and solve the error. I don't know if it could be realted too python version because I've python3.9 and in another system that I used to test it with python3.12 it worked fine.

In python 3.9 system the issue was with the code in line 53 of encryption.py file

`public_key = serialization.load_pem_public_key(
            key_data['pubKey'].encode(),
            backend=default_backend
 )`

Python considers that you are setting **backend** variable with another variable called **default_backend** but **defacult_backend** is insted a function. So you have to add parenthesis to solve it. The final code should be:

`public_key = serialization.load_pem_public_key(
            key_data['pubKey'].encode(),
            backend=default_backend()
        )`

Thanks
Kind regards,

jgriss, 2025-03-15:
Hi! This is now fixed in v0.0.27

closed 2025-03-15 by jgriss