jgriss/FusionSolarPy #32: The individual values ​​of the optimizers are not displayed

RRonGit, 2024-10-16:
The individual values ​​of the optimizers are not displayed, when I run the script I see the following:

Traceback (most recent call last):
  File "/pythonskripts/huawei.py", line 50, in <module>
    for x in client.get_optimizer_stats(client.get_device_ids()['Inverter']):
                                        ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'Inverter'

Script huawei.py:

[image]

RRonGit, 2024-10-22:
output is: {'Optimizer': 'NE=xxxxxxxxxx'}

Note:
I have hardcoded plant_ids = ["NE=xxxxxxxxxx"] in the script

jgriss, 2024-10-22:
Based on this, you'd have to adapt the code as follows:

```python3
for x in client.get_optimizer_stats(client.get_device_ids()['Optimizer']):
  print(x)
```

RRonGit, 2024-10-24:
Thanks for the tip, unfortunately I still don't get any data from the optimizers with the change.