jgriss/FusionSolarPy #38: Issue with Authentication

BCosta28, 2025-02-24:
Hello! This has been an extremely useful tool for me in these past weeks, but i have recently encountered a problem that stops me from logging in to certain accounts through my script.

For context, i have access to 6 different Fusion Solar accounts, and by using this code:

```
def fusion_login(id, password, subdomain):
    client = client = FusionSolarClient(
        id,
        password,
        huawei_subdomain = subdomain
    )
```

I can access two of them. The rest of them return this:

`fusion_solar_py.exceptions.AuthenticationException: Failed to login into FusionSolarAPI: Login failed. Enter the correct username and password.`

But the problem is that my login is correct, since i can access all of the accounts without any problem through the online platform.

I am trying to develop a script that can fectch data from all the accounts I have access too, and this problem is preventing me from continuing this project. Any help you could give me would be very appreciated!

Thank you for you time!

BCosta28, 2025-02-24:
I managed to fix it. For some reason, typing the password or copying it from my list gave different outputs. But now i have another problem. When i login with two of the accounts, after the login it asks me to select one of two regions.

[image]

When my code tries to login into this account, i get this error:

`Using login subdomain: eu5
Login Response JSON: {'errorCode': '470', 'errorMsg': None, 'redirectURL': None, 'respMultiRegionName': ['region004', 'region05'], 'serviceUrl': None, 'verifyCodeCreate': False, 'twoFactorStatus': None}
Traceback (most recent call last):
  File "C:\Users\bruno\AppData\Local\Programs\Python\Python311\Lib\site-packages\urllib3\connection.py", line 169, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bruno\AppData\Local\Programs\Python\Python311\Lib\site-packages\urllib3\util\connection.py", line 73, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bruno\AppData\Local\Programs\Python\Python311\Lib\socket.py", line 962, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
socket.gaierror: [Errno 11001] getaddrinfo failed`

Any idea on how to fix this? Thank you!

jgriss, 2025-03-15:
Hi @BCosta28 

I wasn't aware that multi-region accounts exist. Since my account is linked to a single region that's nothing I can test.

Did you try to adapt the `huawei_subdomain` parameter for these accounts?