jgriss/FusionSolarPy #40: get_optimizer_stats doesn't work

jgonzalezzitu, 2025-03-12:
I'm trying to see the information about the inverter, because I need the temperature of it (if it's possible also the historical temperature, but I don't find it in the official page). But when I execute the method it returns me an error. I'm trying different ways to call it, but has someone reached that information or knows why it gives me an error? (I tried with different NE numbers, not only the inverters one, if it could be that, but with all of it gives the same error)

jgriss, 2025-03-15:
Hi @jgonzalezzitu 

Thanks a lot for reporting this!

Unfortunately, I do not have an optimizer, therefore I cannot test these functions.

Could you maybe post the code you use, as well as the stack trace of the error you receive?

Thanks a lot!

jgonzalezzitu, 2025-03-18:
Hi @jgriss 

Yes, this is the code I'm using in my tests:

`client = FusionSolarClient(
                email, 
                password,
                huawei_subdomain= "uni003eu5",
                captcha_model_path="C:\\VS2022\\Ores\\bin\\Debug\\net8.0\\python\\captcha_huawei.onnx"
                )

            daysAgo = datetime.datetime.now(datetime.UTC) - datetime.timedelta(days=days)
            
            start = daysAgo.strftime("%Y-%m-%d 00:00:00")
            struct_time = time.strptime(start, "%Y-%m-%d %H:%M:%S")
            seconds = round(time.mktime(struct_time) * 1000)
            optimizer_stats = client.get_optimizer_stats(inverter_id="NE=XXXXXX") #Here the NE of the inverter
`

And there, when I try to execute the get_optimizer_stats, it goes to the method and it doesn't return 'succes' or 'data'. This is what it returns:

'exceptionType' = 'ROA_EXFRAME_EXCEPTION'
'exceptionId' = 'com.huawei.bsp.roa.unknow.exception'

closed 2025-08-24 by jgriss