jgriss/FusionSolarPy #45: get_plant_stats does not follow local time

andbiz, 2025-06-10:
Hi!

First of all, thanks a lot for this amazing package.

I am using it on a client on UTC +2 (CEST) and I noted that when I run the `get_plant_stats` between 12 AM - 2 AM it downloads the plant data table of the day before. It downloads the correct table only after 2 AM, so (apparently) when a new day starts on UTC time.

Or am I using it in the wrong way?

I attach the data I downloaded today (2025-06-10)  at 01.00.56 AM (but it is the same data I got from 00.05.33 AM to 01.56.18 AM, downloaded every 5 mins)

[2025-06-10_01:00:56.csv](https://github.com/user-attachments/files/20670352/2025-06-10_01.00.56.csv)

Thanks a lot!

jgriss, 2025-06-11:
Hi @andbiz 

Thanks a lot for the feedback!

I'm aware of this issue but unfortunately have not yet found a solution for it. There are several functions that react the same way.

On the one hand, the API seems to be using the set timezone (which is hardcoded to CET at the moment) but when the day changes, it somehow expects UTC.

In case you want to play with the timezone settings, you could use the following code:
```
if not query_time:
  # this function could also be part of the issue...
  query_time = your_client._get_day_start_sec()

# expects that you already logged in and your client is called "your_client"

r = your_client._session.get(
            url=f"https://{self._huawei_subdomain}.fusionsolar.huawei.com/rest/pvms/web/station/v1/overview/energy-balance",
            params={
                "stationDn": plant_id,
                "timeDim": 2,
                "queryTime": query_time, 
                "timeZone": 2,  # 1 in no daylight
                "timeZoneStr": "Europe/Vienna",
                "_": round(time.time() * 1000),
            },
        )
r.raise_for_status()
plant_data = r.json()

# return the actual data
actual_data = plant_data["data"]
```

andbiz, 2025-06-13:
Thanks @jgriss , I will keep testing it and let you know if somethings comes out.

closed 2025-11-22 by jgriss