jgriss/FusionSolarPy #46: Reset the session parameter in the logged_in decorator

gaat21, 2025-08-03:
Hi

First of all this package is very useful. 
I have an problem with it. I would like to use the client with my session but this session will be overridden in the logged_in decorator.

Here:  https://github.com/jgriss/FusionSolarPy/blob/289843710798878f15b4e2bd2a8da26433689622/src/fusion_solar_py/client.py#L162

Is there any chance to fix this issue?

jgriss, 2025-08-24:
Hi @gaat21 

This code is only called if the session is no longer active - ie. we need to log in again. In the constructor, the passed session is used and it is used as long it provides access to the API.

If the connection is used over a long period of time, the session sometimes becomes invalid and we have to reset it. Unfortunately, re-creating the session object is the easiest way to achieve this.

You can access the current session object through the `your_client._session` variable.

Do you have a different use-case in minde?

jgriss, 2025-11-22:
Hi @YKPDang 

Thanks a lot for the suggestions! I followed your suggestions and changed session reuse in version 0.1.0